pulp-platform/pulpino #373: Unable to generate bitstream for Arty A7 35t

SubhamRath, 2021-03-16:
Hello, we are trying to implement pulpino on Arty a7 35t but it is showing the following error. Can anyone please help us with this?

``````
source pulpino.tcl -notrace
CRITICAL WARNING: [Board 49-4] Problem parsing board_part file - /home/vlsi_lab/Desktop/project_neumann/Vivado_2015.1/Vivado/2015.1/data/boards/board_files/genesys-zu-3eg/B.0/board.xml, failed to parse board part file. Board Fpga file does not have valid entry for family/vendor/board_name/board_revision.
WARNING: [Board 49-26] cannot add Board Part digilentinc.com:arty-s7-25:part0:1.0 available at /home/vlsi_lab/Desktop/project_neumann/Vivado_2015.1/Vivado/2015.1/data/boards/board_files/arty-s7-25/E.0/board.xml as part xc7s25csga324-1 specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part digilentinc.com:arty-s7-50:part0:1.0 available at /home/vlsi_lab/Desktop/project_neumann/Vivado_2015.1/Vivado/2015.1/data/boards/board_files/arty-s7-50/B.0/board.xml as part xc7s50csga324-1 specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part digilentinc.com:cmod-s7-25:part0:1.0 available at /home/vlsi_lab/Desktop/project_neumann/Vivado_2015.1/Vivado/2015.1/data/boards/board_files/cmod-s7-25/B.0/board.xml as part xc7s25csga225-1 specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part digilentinc.com:cora-z7-07s:part0:1.0 available at /home/vlsi_lab/Desktop/project_neumann/Vivado_2015.1/Vivado/2015.1/data/boards/board_files/cora-z7-07s/B.0/board.xml as part xc7z007sclg400-1 specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part digilentinc.com:genesys2:part0:1.1 available at /home/vlsi_lab/Desktop/project_neumann/Vivado_2015.1/Vivado/2015.1/data/boards/board_files/genesys2/H/board.xml as part xc7k325tffg900-2 specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part digilentinc.com:sword:part0:1.0 available at /home/vlsi_lab/Desktop/project_neumann/Vivado_2015.1/Vivado/2015.1/data/boards/board_files/sword/C.0/board.xml as part xc7k325tffg900-2 specified in board_part file is either invalid or not available
Command: synth_design -top pulpino -part xc7a35ticsg324-1L -directive AreaOptimizedHigh -control_set_opt_threshold 1
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xc7a35ti'
WARNING: [Common 17-348] Failed to get the license for feature 'Synthesis' and/or device 'xc7a35ti'
0 Infos, 7 Warnings, 1 Critical Warnings and 1 Errors encountered.
synth_design failed
ERROR: [Common 17-345] A valid license was not found for feature 'Synthesis' and/or device 'xc7a35ti'. Please run the Vivado License Manager for assistance in determining
which features and devices are licensed for your system.
Resolution: Check the status of your licenses in the Vivado License Manager. For debug help search Xilinx Support for "Licensing FAQ". 
INFO: [Common 17-206] Exiting Vivado at Tue Mar 16 12:39:30 2021...
[Tue Mar 16 12:39:35 2021] synth_1 finished
wait_on_run: Time (s): cpu = 00:00:00.14 ; elapsed = 00:00:12 . Memory (MB): peak = 1491.066 ; gain = 0.000 ; free physical = 246 ; free virtual = 3917
# open_run synth_1
ERROR: [Common 17-69] Command failed: Run 'synth_1' failed. Unable to open
INFO: [Common 17-206] Exiting Vivado at Tue Mar 16 12:39:35 2021...
Makefile:5: recipe for target 'all' failed
make[1]: *** [all] Error 1
make[1]: Leaving directory '/home/vlsi_lab/Desktop/project_neumann/RISC-V/pulpino/fpga/pulpino'
Makefile:39: recipe for target 'pulpino/pulpino.edf' failed
make: *** [pulpino/pulpino.edf] Error 2
``````

Iamcaibirds, 2021-03-16:
It seems that you do not have license for vivado

SubhamRath, 2021-03-17:
I am currently using the webpack version, so I guess license should not be an issue?

Iamcaibirds, 2021-03-17:
you can try another version of vivado which needs a license, and check whether the issue still there.

babekmemmedov1985, 2021-04-27:
I have license but it have the same problem

Iamcaibirds, 2021-04-27:
@babekmemmedov1985 do you use the method in the readme file to generate the whole project?