pulp-platform/pulpino #376: Error loading design(make hello world.vsim)

fanyiyang1, 2021-06-14:
Hello, I met some problem when I "make hello world.vsim"
The message in ModelSim console shows :
source tcl_files/run.tcl
# ** Warning: (vsim-159) Mode option -64 is not supported in this context and will be ignored.
# ** Error: (vsim-3281) Invalid command line generic assignment "-gRISCY_RV32F=".
# Use the -help option for complete vsim usage.
# vsim -quiet tb -L pulpino_lib -L axi_node_lib -L apb_node_lib -L axi_mem_if_DP_lib -L axi_spi_slave_lib -L axi_spi_master_lib -L apb_uart_sv_lib -L apb_gpio_lib -L apb_event_unit_lib -L apb_spi_master_lib -L fpu_lib -L apb_pulpino_lib -L apb_fll_if_lib -L core2axi_lib -L apb_timer_lib -L axi2apb_lib -L apb_i2c_lib -L zero_riscy_lib -L axi_slice_dc_lib -L riscv_lib -L apb_uart_lib -L axi_slice_lib -L adv_dbg_if_lib -L apb2per_lib "+nowarnTRAN" "+nowarnTSCALE" "+nowarnTFMPC" "+MEMLOAD=PRELOAD" -t ps -voptargs="+acc -suppress 2103" -GTEST="" -gRISCY_RV32F= -gZERO_RV32E= -gZERO_RV32M= -gUSE_ZERO_RISCY= 
# Error loading design
# Errors: 1, Warnings: 1
# 

I am not sure whether it's the problem with my vsim path. I write the vsim path in the '.sh' file: VSIM=/home/luckyyf/Desktop/MT2021/modelsim_ase/bin

When I "make vcompile", the output is :  --> PULPino compilation complete! 
--> Compiling work.tb... 
Compiling component:  work.tb 

--> work.tb compilation complete! 

--> PULPino platform compilation complete! 

Built target vcompile

And the output of command "make helloworld.vsimc VERBOSE=1":
/usr/local/bin/cmake -S/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build -B/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
make  -f CMakeFiles/Makefile2 helloworld.vsim
make[1]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
/usr/local/bin/cmake -S/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build -B/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/bin/cmake -E cmake_progress_start /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/CMakeFiles 4
make  -f CMakeFiles/Makefile2 apps/helloworld/CMakeFiles/helloworld.vsim.dir/all
make[2]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
make  -f libs/bench_lib/CMakeFiles/bench.dir/build.make libs/bench_lib/CMakeFiles/bench.dir/depend
make[3]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
cd /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/libs/bench_lib /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/libs/bench_lib /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/libs/bench_lib/CMakeFiles/bench.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
make  -f libs/bench_lib/CMakeFiles/bench.dir/build.make libs/bench_lib/CMakeFiles/bench.dir/build
make[3]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
make[3]: Nothing to be done for 'libs/bench_lib/CMakeFiles/bench.dir/build'.
make[3]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
[  0%] Built target bench
make  -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/depend
make[3]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
cd /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/CMakeFiles/crt0.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
make  -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/build
make[3]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
make[3]: Nothing to be done for 'CMakeFiles/crt0.dir/build'.
make[3]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
[ 25%] Built target crt0
make  -f libs/string_lib/CMakeFiles/string.dir/build.make libs/string_lib/CMakeFiles/string.dir/depend
make[3]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
cd /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/libs/string_lib /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/libs/string_lib /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/libs/string_lib/CMakeFiles/string.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
make  -f libs/string_lib/CMakeFiles/string.dir/build.make libs/string_lib/CMakeFiles/string.dir/build
make[3]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
make[3]: Nothing to be done for 'libs/string_lib/CMakeFiles/string.dir/build'.
make[3]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
[ 25%] Built target string
make  -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/depend
make[3]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
cd /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/libs/sys_lib /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/libs/sys_lib /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/libs/sys_lib/CMakeFiles/sys.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
make  -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/build
make[3]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
make[3]: Nothing to be done for 'libs/sys_lib/CMakeFiles/sys.dir/build'.
make[3]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
[ 50%] Built target sys
make  -f apps/helloworld/CMakeFiles/helloworld.elf.dir/build.make apps/helloworld/CMakeFiles/helloworld.elf.dir/depend
make[3]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
cd /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/apps/helloworld /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/apps/helloworld /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/apps/helloworld/CMakeFiles/helloworld.elf.dir/DependInfo.cmake --color=
[... 27 lines cut ...]
make[3]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
make  -f apps/helloworld/CMakeFiles/helloworld.slm.cmd.dir/build.make apps/helloworld/CMakeFiles/helloworld.slm.cmd.dir/build
make[3]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
make[3]: Nothing to be done for 'apps/helloworld/CMakeFiles/helloworld.slm.cmd.dir/build'.
make[3]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
[100%] Built target helloworld.slm.cmd
make  -f apps/helloworld/CMakeFiles/helloworld.vsim.dir/build.make apps/helloworld/CMakeFiles/helloworld.vsim.dir/depend
make[3]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
cd /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/apps/helloworld /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/apps/helloworld /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/apps/helloworld/CMakeFiles/helloworld.vsim.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
make  -f apps/helloworld/CMakeFiles/helloworld.vsim.dir/build.make apps/helloworld/CMakeFiles/helloworld.vsim.dir/build
make[3]: Entering directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
[100%] Running helloworld in ModelSim
[... cut ...]
Couldn't load font /home/luckyyf/Desktop/MT2021/modelsim_ase/linuxaloem/VisualizerRls/fonts
Reading pref.tcl
make[3]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
[100%] Built target helloworld.vsim
make[2]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'
/usr/local/bin/cmake -E cmake_progress_start /home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build/CMakeFiles 0
make[1]: Leaving directory '/home/luckyyf/Desktop/MT2021/MT/pulpino/sw/build'

Could anyone kindly tell me how to solve it? Thank you!

sunil3690, 2021-06-15:
hi,
fix your modelsim PATH and run.

Regards,

Sunil Rathod

fanyiyang1, 2021-06-15:
> hi, fix your modelsim PATH and run. Regards, Sunil Rathod
> […](#)

Thanks for your information.
Where can I fix the modelsim Path? You mean in the cmake_configure-- "VSIM"?
I have tried VSIM=/home/luckyyf/Desktop/MT2021/modelsim_ase/bin, 
VSIM=/home/luckyyf/Desktop/MT2021/modelsim_ase/bin/vsim,
VSIM=/home/luckyyf/Desktop/MT2021/modelsim_ase/linuxaloem...
But they all failed... Do you have any ideas?
I am new in this field and maybe my questions are basic. Sorry for that.

sunil3690, 2021-06-15:
The Path i have fixed was like this
....../root/pulpino/altera/16.0/modelsim_ase/linuxaloem
and need to remove -64 from CMAKE files.

Regards,

Sunil Rathod

fanyiyang1, 2021-06-15:
> The Path i have fixed was like this ....../root/pulpino/altera/16.0/modelsim_ase/linuxaloem and need to remove -64 from CMAKE files. Regards, Sunil Rathod
> […](#)

**Oh Yes I have changed my path to ……/modelsim_ase/linuxaloem
But this time when I make helloworld.vsim, the output becomes:**
[  0%] Built target bench
[ 25%] Built target crt0
[ 25%] Built target string
[ 50%] Built target sys
[ 75%] Built target helloworld.elf
[ 75%] Generating vectors/stim.txt
[ 75%] Built target helloworld.stim.txt
[ 75%] Built target helloworld.links
[100%] Built target helloworld.slm.cmd
[100%] Running helloworld in ModelSim
**env: ‘/home/luckyyf/Desktop/MT2021/modelsim_ase/linuxaloem’: Permission denied**
apps/helloworld/CMakeFiles/helloworld.vsim.dir/build.make:76: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim' failed
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsim] Error 126
CMakeFiles/Makefile2:4110: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/all' failed
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/all] Error 2
CMakeFiles/Makefile2:4117: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule' failed
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
Makefile:668: recipe for target 'helloworld.vsim' failed
make: *** [helloworld.vsim] Error 2

**I don't know where to delete '-64' you mentioned... I can't find it in CMakelist, CMakeCache or cmake_configure...
Even I use the sudo make... It still shows that env: ‘/home/luckyyf/Desktop/MT2021/modelsim_ase/linuxaloem’: Permission denied?
Thanks very much!**

sunil3690, 2021-06-15:
please go through all cmake files including cmake simulation etc.

Regards,

Sunil Rathod

sunil3690, 2021-06-15:
I have even worked on the pulpino in past years. I did not remember the
exact steps.

Regards,

Sunil Rathod

fanyiyang1, 2021-06-15:
> I have even worked on the pulpino in past years. I did not remember the exact steps. Regards, Sunil Rathod On Tue, Jun 15, 2021 at 3:07 PM Sunil Rathod ***@***.***> wrote:
> […](#)

Okay I will have a try. Thanks for your kind reply: )!

closed 2021-06-25 by fanyiyang1

haresh-seenivasagan, 2024-01-08:
HI @fanyiyang1 were you able to if so could you guide me on how i can resolve this as well ? I am encountering the same issue and changing the path hasnt helped me.
i am also unsure about where to delete the -64 from CMAKE files.