pulp-platform/pulpino #396: There is no input signal in modelsim

AdmsLCW, 2023-04-04:
I have successfully printed hello world but I can't observe the input signal and the signal  I want because of optimization, and when I turn off optimization, modelsim doesn't work, how can I fix this.
The modelsim version is modelsim-se-2020.4

wmhaikal00, 2023-04-19:
Hi AdmsLCW

May I know how did you to compile the RTL using the modelsim-se-2020.4 ? I'm trying to compile, however I met with no rule to make target error.

AdmsLCW, 2023-04-26:
No target error may be happened in cmake，you should check the version of cmake.

AdmsLCW, 2023-04-26:
In order to solve this problem, I run the pulpino by vcs. And now I can see the waveform which i need.

closed 2023-04-27 by AdmsLCW

sharad2000, 2025-04-16:
Hi @AdmsLCW actually I am not able to get Hello world, and I am also using vcs. Can you Please tell me how I can debug this?
Thanks

AdmsLCW, 2025-04-16:
Emm, it's been so long that I forgot the details. First, make sure the file is complete, then modify the header file path, or load it in the script. Finally, compile the files in the order of startup in the original project and start the simulation. Maybe I will upload the vcs version of the environment later when I'm free.

sharad2000, 2025-04-16:
Hi @AdmsLCW actually I have done all of the required steps which were given in the readme part, and I have also generated the .slm files which were needed for instruction memory data and data memory data for Helloworld.c code.

Now I am simulating it and also gotten the waveform through verdi but I am just not able to find out what is the issue is as I am not able to see hello world.

I sincerely request you to please share any possible data that you can at your earliest convenience.
Thank you so much for your help and support.

AdmsLCW, 2025-04-16:
Did you save the output content to a file？

sharad2000, 2025-04-16:
@AdmsLCW 

After running ./simv command this is showing 

Using MEMLOAD method: PRELOAD
Using      ri5cy core
*Verdi* Loading libsscore_vcsmx202409.so
FSDB Dumper for VCS-MX, Release Verdi_W-2024.09-SP2, Linux x86_64/64bit, 03/03/2025
(C) 1996 - 2025 by Synopsys, Inc.
*Verdi* : Create FSDB file 'result.fsdb'
*Verdi* : Begin traversing the scope (tb), layer (0).
*Verdi* : End of traversing.
*Verdi* : fsdbDumpon - All FSDB files at 0 ps.
[SPI] Enabling QPI mode
[adv_dbg_if] AXI4 WRITE         32 burst @1a107008 for           4 bytes.
Preloading memory
Preloading instruction memory from slm_files/l2_stim.slm
Preloading data memory from slm_files/tcdm_bank0.slm
[TRACER] Output filename is: trace_core_00_0.log
RX string:                                                                                                                                                                                                                                                                 
[SPI] Received 00000000
[SPI] Test OK

sharad2000, 2025-06-12:
Hi @AdmsLCW 

I am currently working on extending the PULPino platform and integrating it with additional IPs, including a UART module that adheres strictly to the AMBA APB protocol specification. During this process, I noticed a protocol compliance issue in the existing APB implementation within PULPino, specifically in the way the UART peripheral is accessed.

In the current PULPino design, the UART interface appears to operate using only the Access phase of the APB protocol (i.e., PSEL=1, PENABLE=1). The Setup phase (PSEL=1, PENABLE=0), which is an essential part of the APB specification, is not utilized or enforced.
Because of this, the peripheral logic directly reacts to PSEL and PENABLE both being high, without validating if the correct transition from the setup to access phase has occurred.

Read and write operations may not behave as expected, especially when used with verified third-party IPs.

A possible solution might be to Introduce a simple FSM or state-tracking logic at the peripheral interface level that distinguishes between setup and access phases.

Please let me know your thoughts on this, and whether such enhancements would be considered for future updates.

Thank you for listening.