pulp-platform/pulpino #397: make vcompile: No rule to make target

wmhaikal00, 2023-04-19:
Hello and hi everyone ! 

I'm trying to compile the RTL. However, i met with this issue. Can anyone advice me on what should i do to compile the RTL successfully ? Your help is much appreciated

[image]

AdmsLCW, 2023-04-27:
Select a configuration file about riscv, put it in the build folder, and run the configuration file in the terminal (because the file is executable). When the configuration is complete, enter make vcompile.

wmhaikal00, 2023-05-03:
Can you give me the specific name of the configuration file? I don't quite get it. Thanks!

wmhaikal00, 2023-05-04:
[image]

closed 2023-05-04 by wmhaikal00

wmhaikal00, 2023-05-04:
I tried to run the cmake_configure.riscv.gcc.sh as you suggested. But it seems the system is unknown to cmake. Thus it unable to compile simple test program. My current version of cmake is 3.10.2. 

[image]

Do you have any suggestions on how should i fix this problem ?

AdmsLCW, 2023-05-04:
The problem is that you do not have the cross-compiler installed, and the configuration file cannot find the cross-compiler. You need to download the cross-compiler (https://github.com/pulp-platform/ri5cy_gnu_toolchain). If you have installed the cross-compiler and still cannot find the cross-compiler, you just did not add the path of the cross-compiler to the environment variable in the user. If you have trouble compiling the cross-compiler, try compiling with gcc version 5.2 or 5.20 (I don't remember exactly).