pulp-platform/pulpino #399: An Error occurred when running update-ips.py

YanWon, 2023-06-23:
Other IPs was correctly downloaded to the local folder, the adv_dbg_if IP was downloaded too, however an error was showed. Does this error will influence the running of the whole project?

SUMMARY
    adv_dbg_if - Could not checkout commit pulpinov1
()
ERRORS during IP update!

By the way, forgive my poor English.Thanks!

ConstanceMitel, 2023-07-28:
I got the same error with update-ips and when I tried to run make vcompile I got the following error:

_Compiling component: adv_dbg_if

** Error: (vlog-7) Failed to open design unit file "../ips/adv_dbg_if/rtl/adbg_lint_biu.sv" in read mode.
No such file or directory. (errno = ENOENT)

make[3]: *** [CMakeFiles/vcompile] Error 1
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
make: *** [vcompile] Error 2_

Did you get the same? Did you manage to fix it?

ericliu021, 2023-08-17:
Don't worry, I also got the same error after runing update-ips.py, it seems the verilator-pulpino branch of adv_dbgif module has been deleted in latest update, but this doesn't affect the compiler or simulation work in later.

ConstanceMitel, 2023-08-22:
> Don't worry, I also got the same error after runing update-ips.py, it seems the verilator-pulpino branch of adv_dbgif module has been deleted in latest update, but this doesn't affect the compiler or simulation work in later.

But it does affect it. You cannot run make vcompile, because it doesn't find the necessary files. 
Unless there is a workaround for this problem, through editing the scripts, the classic workflow of pulpino as described, does not work anymore.

ericliu021, 2023-08-22:
> But it does affect it. You cannot run make vcompile, because it doesn't find the necessary files. Unless there is a workaround for this problem, through editing the scripts, the classic workflow of [...]

> I got the same error with update-ips and when I tried to run make vcompile I got the following error:
>
> _Compiling component: adv_dbg_if
>
> ** Error: (vlog-7) Failed to open design unit file [...]

Actually I can find adbg_lint_biu.sv file under adv_dbg_if and in github, you can also find this file in matser branch, you can check it in this link(https://github.com/pulp-platform/adv_dbg_if/tree/master/rtl).

rbevin777, 2024-02-29:
Hello, the problem here is that the branch that script looks for is deleted. This script uses `ips_list.yml` to checkout a certain commit namely `pulpinov1`. The problem is that for `adv_dbg_if` is that this commit has been deleted, so the fix (for now at least) is to checkout the last commit where pulpinov1 was merged.

TLDR; Update the ips_list.yml and replace this:
```
adv_dbg_if:
  commit: pulpinov1
```
with this:
```
adv_dbg_if:
  commit: c440ae850a680a2962edf142c8d7e52c3cfed221
```