pulp-platform/pulpino #401: riscv32-unknown-elf-gcc: error: '-march=RV32IM': ISA string must begin with rv32 or rv64

abeitian, 2023-12-15:
The following error occurs when executing make helloworld.vsim:

[  0%] Building C object libs/bench_lib/CMakeFiles/bench.dir/src/bench.c.o
riscv32-unknown-elf-gcc: error: '-march=RV32IM': ISA string must begin with rv32 or rv64
make[3]: *** [libs/bench_lib/CMakeFiles/bench.dir/build.make:63：libs/bench_lib/CMakeFiles/bench.dir/src/bench.c.o] error

This is a bit confusing. I tried changing the value of -march, but it didn't work.

abeitian, 2023-12-18:
Ah, this is a very simple problem. It can be resolved by correctly using the installation method for [Installation (Pulp)](https://github.com/pulp-platform/pulp-riscv-gnu-toolchain?tab=readme-ov-file#installation-pulp) within [pulp](https://github.com/pulp-platform/pulp-riscv-gnu-toolchain). However, initially, I overlooked the fact that a toolchain already existed in the virtual machine, which led to me not using the newly compiled version.

closed 2023-12-18 by abeitian

shane1604, 2024-05-11:
Hi, can u explain more? Im in the same error. And I had riscv-gnu-toolchain

sharad2000, 2025-03-17:
Hi 
Did you solve this error? can you please explain the solution?